sh123/esp32_loradv #3: i make a sch for loradv ,and i tested for heltec-v2.1

jumbo5566, 2023-03-01:
i make a sch for loradv ,and i tested for heltec-v2.1 board ,the default config tx mic ,rx spk sould is not big engouht,the sound quality is not good,but we can try more.

i nottice meshtastic module have codec2 sound tx rx function now.
it just use 4 gpio to work
I2S Word Select
The GPIO to use for the WS signal in the I2S interface.
I2S Data IN
The GPIO to use for the SD signal in the I2S interface.
I2S Data OUT
The GPIO to use for the DIN signal in the I2S interface.
I2S Clock
The GPIO to use for the SCK signal in the I2S interface.
https://meshtastic.org/docs/settings/moduleconfig/audio

sh123, 2023-03-01:
Encoder should change the volume when you rotate it clockwise, quality depends on the codec2 mode, 700 codec2 mode cannot have good sound quality in principle.

jumbo5566, 2023-03-02:
for Good loradv we can view this project:
https://m17project.org/get-started/software

jumbo5566, 2023-03-02:
https://github.com/nakhonthai/M17AnalogGateway
it is esp32 work wifi and Analog Codec2
i tested work good.
but it is only wifi walkie talkie.

sh123, 2023-03-03:
> for Good loradv we can view this project: https://m17project.org/get-started/software

not sure, it requires some effort to implement m17 protocol support

closed 2023-04-13 by sh123