sh123/esp32_loradv #7: add Reliable Packets for voice packet

jumbo5566, 2023-04-27:
hi,friend,
i 'm testing digital voice,and digital photo for lora
i 'm try to add Reliable-Packets for loradv,then we can sent digital packet in Reliable mode,this can increat distance
and voice quality.
https://stuartsprojects.github.io/2021/08/25/Reliable-Packets-Part1.html

sh123, 2023-04-27:
not sure that acknowledging each voice packet is good idea for the same reasons why RTP over UDP is commonly used in Internet

arodland, 2023-04-27:
Yeah, adding ARQ to a realtime stream like voice makes things worse, not better.

LoRa already has FEC, and the default config for esp32_loradv already uses a rate 4/7 code, which is close to the maximum. If the packet doesn't get through even with that, it doesn't make sense to retransmit — you probably don't have *time* for a retransmission before the next voice packet needs to go out. If you *do* have time for a retransmission, then you could equally well go one SF step higher, putting twice as much energy into each bit, and get more range that way.

Also, using reliable mode totally breaks the usecase where there is one transmitter and multiple receivers, which is very common for simplex voice.

closed 2023-05-05 by sh123