sh123/esp32_loradv #17: New PCB ISP programming unsuccessful

Radiomix2000, 2025-01-16:
Tried to program ESP32 soldered to PCB with no success. Every time it terminated with status: "Failed to connect to ESP32: Download mode successfully detected, but getting no sync reply: The serial TX path seems to be down".
Programming ESP32 board outside PCB works good.
How to program ESP32, when it's already soldered to PCB?

sh123, 2025-01-16:
Never had this issue, it should work unless you soldered GPS module or your ESP32 pinouts are different.

Radiomix2000, 2025-01-16:
Yes? I  have GPS module soldered onto PCB. Then it could be the problem. Will try to sort it off.

sh123, 2025-01-16:
Do not solder it, sorry, I did not mention that, I just added place for GPS module for future experiments, so that I do not need to order new boards. There were no free pinouts anymore so I wired them to TX/RX, it should work, but programming and USB usage is not possible.

closed 2025-01-16 by sh123